hz12opensource/libresdr #1: How to populate firmware to QSPI flash (internal) ?

kovalroma, 2024-01-09:
Hello,

Could you please provide more details on how to write firmware to the internal flash drive? I've successfully run your firmware from the SD Card and now want to transfer it to the QSPI flash.

In your instructions, you mentioned:

> Once it is running, the software can be updated using a mounted drive, just like with PlutoSDR. This will populate the QSPI flash, and LibreSDR will be able to run without an SD Card, with USB OTG acting as the normal PlutoSDR.

However, I am having trouble with DFU not working in my case. Could you please direct me to some documentation for LibreSDR?

Thanks!

hz12opensource, 2024-01-15:
Thank you for the question, and glad to know the firmware works for you!

Actually I never tried the reprogram the SPI flash, so I am not sure about the procedure. I assume the standard procedure of PlutoSDR should work similarly. Unfortunately I don't have time recently to test this out.

If you figure it out, it will be great if you can update the instructions on the README page and submit a pull request. Your contribution will be greatly appreciated.

bbjunkie, 2024-01-15:
I have successfully programmed the custom image to QSPI recently. From memory I did the following; 

1) Write the contents of build_sdimg to the root of SD card. 
2) Boot the Libre SDR with USB cable in the left hand socket of the radio
(In windows) the radio is automatically then recognised as a flashdrive, Drop the contents of build_sdimg into the root of the "flashdrive"
Power down the radio, remove the SD card and power up. 
Done

Thanks once again to those who have provided the custom image. 

73 Pete Mi0VAX

kovalroma, 2024-01-18:
@hz12opensource  I found that to be able to populate the QSPI flash, I need both build_sdimg as well as frm files. Could you please also attach frm files to release section ?

hz12opensource, 2024-01-21:
@kovalroma I uploaded the libre.frm and boot.frm here [firmware.tar.gz](https://github.com/hz12opensource/libresdr/files/13999849/firmware.tar.gz) for your testing. It was built using non-overclocking firmware. However I have not got a chance to test them and cannot guarantee them to work. If you manage to get success, I can try to compile other overclocking firmwares.

Quarmire, 2024-01-25:
@hz12opensource Thank you for you work on this firmware!  Correct me if I'm wrong, but I don't think official PlutoSDR firmware works with the LibreSDR.  Unfortunately, I recently messed up my LibreSDR's u-boot somehow and have only got my device to enter DFU mode by flashing your boot.frm.  The SDR shows up as "Analog Devices, Inc. USB download gadget" which I can then use dfu-util to attempt flashing PlutoSDR's official DFU firmware.  Would you happen to have dfu files I could use to attempt a recovery?

hz12opensource, 2024-01-26:
@Quarmire You can boot from SD Card and flash the original firmware back.

You can take a look at /sbin/update.sh, which contains the actual update procedure. It will be executed once you copy firmware to the virtual "flashdrive". The firmware update was done using some `dd` commands to `/dev/mtdblockX`
https://github.com/hz12opensource/libresdr/blob/380307610ba56403268c664afddba88bc9d326b3/patches/buildroot.diff#L7610

Quarmire, 2024-01-26:
@hz12opensource I utilized all the `dd` commands in the update.sh file.  The firmware seems to have been written to the QSPI flash successfully; however, upon powering off the device and removing the SD card, the SDR shows up as an "Analog Devices, Inc. USB download gadget".  This proves that the firmware flashes properly, but it does not boot properly.  As of now, I can still utilize the SDR via SD card, but if you come up with a solution, then I'm all ears.

alokot, 2024-01-29:
I have the same issue with my LibreSDR. It's working only from SD. But in addition my LibreSDR not working on 192.168.2.1 a
[diagnostic_report.txt](https://github.com/hz12opensource/libresdr/files/14086823/diagnostic_report.txt)
nd looks like some issue with ethernet.

kovalroma, 2024-01-31:
> @kovalroma I uploaded the libre.frm and boot.frm here [firmware.tar.gz](https://github.com/hz12opensource/libresdr/files/13999849/firmware.tar.gz) for your testing. It was built using [...]

Hello, I've checked your firmware. An error occurred.  
```
# update_frm.sh ./libre.frm 
239+1 records in
239+1 records out
Warning: Bad CRC, using default environment
Done
```
Something wrong. 
```
md5sum libre.frm 
075599e6b16630d2826aec8034304666  libre.frm
```

kovalroma, 2024-02-11:
@Quarmire hello,
You can try [this firmware](https://mega.nz/file/X9I3yB4L#Gfm9KyFSobdx5Vl0sknCy8ylg8fhjj82iQrmn74tpmg) 

1. Boot with SD card from this repo
2. Log in to SSH and run 
    ```
    update_frm.sh ./pluto.frm
    ```
    **/pluto.frm** file in archive

3. Then turn off you device. Eject SD card and turn it on. After this it should boot from internal MMC.

Quarmire, 2024-02-12:
Thank you, @kovalroma !
Your solution worked, and the LibreSDR is booting from the QSPI again.  I simply flashed the LibreSDR with `libre.frm` using the mass storage device method one would use for a PlutoSDR.

Antowka, 2024-04-22:
I solved a bit another way:

1) unpack libre_base_release_0.37.zip on sdcard and runed with it

2) When LibreSDR was started from sdcard, I copy from libre_base_release_0.37.zip all files *.frm to pluto virtual disk

3) Eject pluto virtual disk and wait when it started again.

4) When Libre was started I disconnect  power and remove sdcard from device

[libre_base_release_0.37.zip](https://github.com/hz12opensource/libresdr/files/15062803/libre_base_release_0.37.zip)

It's work for me

4Z1ZV, 2025-01-10:
Thanks.  It worked for me as well.

However the blue LED did not blink like in the original Pluto.  Had to just wait...

VAlduinV, 2025-10-26:
Take out the SD card and put BOOT.bin there.