hz12opensource/libresdr #5: LibreSDR not booting from SD-card

RasmusEk, 2024-03-28:
Hi, I realize that you aren't responsible for the LibreSDR but I kinda need help

I tried to copy your pre-built firmware on a freshly formated SD card fileformat FAT32

my LibreSDR does not seem to boot with them and keeps acting as a plutoSDR. 
I draw this conclusion from not beeing able to detect it's ethernet port by pinging it, sniffing it or just seeing it in winbox. It still acts as a Pluto working fine from a USB port however I kinda need the ethernet capability.

I think I might be missing the step "Once it is running, the software can be updated using mounted drive" could you explain this to me in greater detail as I struggle to figure what I am supposed to do

rabubake, 2024-04-09:
Having the same issue. Wondering if anyone has managed to get the Ethernet to work and what steps they took!

rabubake, 2024-04-10:
For those struggling to set this up, make sure your ethernet host and the libresdr are on the same subnet and you should be able to connect. To change the libresdr IP, use the OTG connection and change the config.txt Ethernet USB IP. When using the libresdr, you only need power (I used debug plug) and ethernet (not USB OTG).

DF3AK, 2024-12-01:
I have the same problem. 

What works: 
- Booting without SD-Card from QSPI Flash. I have full functionality and I could update the firmware by copying it into the virtual mass storage device. 
- Booting with stock SD-Card (delivered with libreSDR). On this SD-card, there are two partitions: **BOOT** and **rootfs**. Within **BOOT**, there is a BOOT.BIN, an image.ub, and some more text files and folders, the latter containing definitions for various boards. The partition **rootfs** contains an expanded linux root. 

What does not work:
- Tried to copy the the content from "build_sdimg" to an empty SD-Card (single partition, FAT32). LibreSDR does not boot until I remove the card and it falls back to QSPI flash. All cards are 64GB in size. I tried to resize the partition, does not help.  I used the "overclock_cpu1100_ddr750" as well as "baseclock_cpu750_ddr525" release files.

Note: I do not think one has to erase the QSPI Flash in order to boot from SD. The schematics indicate that, as soon as an SD-Card is inserted, the boot mode is switched from QSPI to SD. 

[image]
[image]

DF3AK, 2024-12-01:
Problem solved.

I followed this [guide](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842385/How+to+format+SD+card+for+SD+boot#HowtoformatSDcardforSDboot-CreatingTwoPartitions) on how to properly prepare a SD-card for the Zynq. Maybe the missing boot flag was the issue.

I propose to update [README.md](https://github.com/hz12opensource/libresdr/blob/main/README.md), because it does not mentioned in detail how to prepare the card. Surprisingly, this [video](https://www.youtube.com/watch?v=HDEegX9IoUg&pp=ygUIbGlicmVzZHI%3D)  explains everything step-by-step, but using just an empty card, formatted  FAT32, did not work for me.

rabubake, 2024-12-02:
Thanks a lot, I have been trying to figure it out for a while! Will try it and update.