hz12opensource/libresdr #6: Maximum Sampling Rate

rabubake, 2024-04-18:
Just wondering if anyone has been able to get 20+ MSPS with their LibreSDRs and any of the precompiled firmwares. I've only been able to max out at ~11.5 MSPS, which is about 2x a regular Pluto. Testing was done using IIO in C running on a PC using iio_buffer_refill at various small and large buffer sizes.

Denisfk1985, 2024-04-19:
Yes, it"s work, I has get 20MHz.bandwith. But not working tcxo correction and external pps and 10MHz reference.

bbjunkie, 2025-01-06:
> Yes, it"s work, I has get 20MHz.bandwith. But not working tcxo correction and external pps and 10MHz reference.

This has just been fixed https://github.com/sdy623/hdl/commit/5b9bcef97dec9202b1c5d43871a1ca2c543c0460