hz12opensource/libresdr #7: Image building error

mrengineer, 2024-05-01:
Hello! Please, help me with my problem

I am using Ubuntu 22.04.4 LTS, Vivado 2022.2 Vitis & Vivado

When I try to do "make" I get

```
*** buffer overflow detected ***: terminated
Aborted (core dumped)
make[1]: *** [Makefile:1187: u-boot] Error 134
make[1]: exit directory «/home/user/libresdr/plutosdr-fw_0.38_libre/u-boot-xlnx»
make: *** [Makefile:76: u-boot-xlnx/tools/mkimage] Error 2
```
Since I have a lot of software intalled, I have prepared docker

```
FROM ubuntu:22.04
RUN apt-get -y update && apt-get -y upgrade
RUN apt-get install -y --no-install-recommends apt-utils git mc libgmp-dev build-essential fakeroot libncurses5-dev libssl-dev ccache u-boot-tools 
RUN apt-get -y install mtools bc python3 cpio zip unzip rsync file 
RUN apt-get -y install wget flex bison language-pack-en libtinfo5 x11-utils xvfb dbus-x11 libswt-glx-gtk-4-jni libgtk2.0-0
RUN DEBIAN_FRONTEND="noninteractive" apt-get -y install libmpc-dev dfu-util device-tree-compiler

WORKDIR /workspace
RUN git clone https://github.com/hz12opensource/libresdr.git
RUN mkdir /tools
```

and docker-compose.yml

```
version: "3.5"

services:
  libre_builder:
    container_name: librsdr_img
    restart: always
    build: .
    image: librsdr_img
    #entrypoint: bash
    volumes:
      - /home/user:/tools
      - .:/workspace
    network_mode: host
    command: bash
```
next in container's cli I do everything and finaly ```make```
Finally, everything is finished OK:
```
...
legal-info/sources/wpa_supplicant-2.10/0002-wpa_supplicant-use-a-less-generic-name-for-IEEE802.1.patch
legal-info/sources/wpa_supplicant-2.10/0001-build-re-enable-options-for-libwpa_client.so-and-.patch
legal-info/sources/wpa_supplicant-2.10/0003-Include-HMAC-SHA384-512-KDF-for-SAE-if-SHA384-512-is.patch
legal-info/sources/wpa_supplicant-2.10/wpa_supplicant-2.10.tar.gz
legal-info/sources/wpa_supplicant-2.10/series
legal-info/sources/libxml2-2.10.4/
legal-info/sources/libxml2-2.10.4/libxml2-2.10.4.tar.xz
rm linux/arch/arm/boot/dts/zynq-libre.dtb
```
and after ```make sdimg``` I see
```
mkdir /workspace/libresdr/plutosdr-fw_0.38_libre/build_sdimg
cp build/sdk/fsbl/Release/fsbl.elf 	/workspace/libresdr/plutosdr-fw_0.38_libre/build_sdimg/fsbl.elf  
cp: cannot stat 'build/sdk/fsbl/Release/fsbl.elf': No such file or directory
make: *** [Makefile:202: sdimg] Error 1
```

**Question: why is fail? What I have to do?**

mrengineer, 2024-05-05:
I guess updates broken something in my host Ubuntu before and it cause problems with compilation and linking while buiilding

Finally, here is the recepie to build:
1. Virtual machine on Virtual box 7 with clean and new Ubuntu 18.04.6 LTS, Desktop version, ~16 RAM, 200-250 GB disk space, 2 CPU (more is better!)
2. No updates for Ubuntu
3. Install Vivado + Vitis + Vitis HLS v. 2021.2 inside VM, installed into ```/opt``` (**it might be important**)
4. Double check that license is active in XIlinx license manager
5. Download repo ver. 0.38
6. ```git clone --branch v0.38 --recursive https://github.com/analogdevicesinc/plutosdr-fw.git plutosdr-fw_0.38_libre```
7. Apply patch for LibreSDR ```./apply.sh```
8. ```cd plutosdr-fw_0.38_libre```
9. Extra and corrected vars
```
export VIVADO_SETTINGS=/opt/Xilinx/Vivado/2021.2/settings64.sh
export TARGET=libre
export VIVADO_VERSION=v2021.2
export ADI_IGNORE_VERSION_CHECK=1
```
10. ```make``` 
11. Wait for a long while ~1..1.5 hours for first time
12. ```make sdimg```

PS: All finished OK, but I will check image on device next few days

mrengineer, 2024-05-06:
Image is good. Closed

closed 2024-05-06 by mrengineer

ryleenhughes, 2024-05-07:
Hello, I'm sorry to disturb you with unrelated question in this issue ticket, but I can't find much activities about this device. 
Would you mind tell can you get the Ethernet port work when using the fw from this repo? if you can, would you mind explain how?

When I connect the LibreSDR to my computer using an Ethernet cable, Windows 10 recognizes it as an "unrecognized network" and displays an obviously incorrect IPv6 address in the detailed view. On Linux, it shows a "connect network fail" message, with limited information available in KDE's network manager.

I have exhausted all the methods I could think of to try to resolve this issue, including:

- Trying all the pre-built binaries available on this GitHub repository, including the base clock and overclocking versions, as well as one I found in the GitHub issue page.
- Setting the IP of the LibreSDR to match the subnetwork range of my local network.
- Setting the IP of the LibreSDR to a completely independent range.
- Attempting various IP configurations.
- Connecting the LibreSDR directly to my router and attempting to establish a connection from my computer.
- Connecting the LibreSDR directly to my computer and attempting to establish a connection.
- Exploring other troubleshooting options and checks.

mrengineer, 2024-05-07:
Ethernet works fine. It has ip address 192.168.2.1

mrengineer, 2024-05-07:
Yor pc must be in same network with  static ip. There's no dhcp on board

ryleenhughes, 2024-05-07:
Thank you so much for your answer!

SO "same network" I assume you meant something like this? 
- connect ether port of LibreSDR to PC
- set one ip between 192.168.2.2 to 192.168.2.254 with netmask 255.255.255.0 for PC

Is that correct? Sorry if my question is stupid

mrengineer, 2024-05-07:
> Thank you so much for your answer!
> 
> SO "same network" I assume you meant something like this?
> 
> * connect ether port of LibreSDR to PC
> * set one ip between 192.168.2.2 to 192.168.2.254 with netmask 255.255.255.0 for PC
> 
> Is that correct? Sorry if my question is stupid

yes

luoyue625, 2024-09-07:
Hello ，Please hellp me with my problem!
I download the soure using "git clone --branch v0.38 --recursive https://github.com/analogdevicesinc/plutosdr-fw.git", while I compile the soure with make, there is no error!
when I contiue to make sdimg and I got the error  as follow
make: *** No rule to make target 'sding'.  Stop.

Then I re-download the soure using cmd "git clone --branch v0.38 --recursive https://github.com/analogdevicesinc/plutosdr-fw.git plutosdr-fw_0.38_libre"
I cannot find the the patch script "./apply.sh"
where is this "apply.sh"?

all the contents as follows:
brs@ubuntu-18:/opt/project/plutosdr-fw_0.38_libre$ ll
total 32
drwxrwxr-x 6 brs brs 4096 9月   7 09:47 ./
drwxrwxr-x 3 brs brs 4096 9月   6 19:43 ../
drwxrwxr-x 2 brs brs 4096 9月   7 09:25 buildroot/
drwxrwxr-x 9 brs brs 4096 9月   7 09:20 .git/
-rw-rw-r-- 1 brs brs  288 9月   6 19:44 .gitignore
-rw-rw-r-- 1 brs brs  420 9月   6 19:44 .gitmodules
drwxrwxr-x 2 brs brs 4096 9月   7 09:25 hdl/
drwxrwxr-x 9 brs brs 4096 9月   7 06:27 plutosdr-fw_0.38_libre/

luoyue625, 2024-09-07:
Hello, pls help me with my problem.
while I download all the source using "git clone --branch v0.38 --recursive https://github.com/analogdevicesinc/plutosdr-fw.git plutosdr-fw_0.38_libre"
I cannot find the script "apply.sh".
all the content as follows:
brs@ubuntu-18:/opt/project/plutosdr-fw_0.38_libre$ find . -name "apply.sh"
brs@ubuntu-18:/opt/project/plutosdr-fw_0.38_libre$ ll
total 72
drwxrwxr-x  9 brs brs 4096 9月   7 19:01 ./
drwxrwxr-x  3 brs brs 4096 9月   7 19:42 ../
drwxrwxr-x 15 brs brs 4096 9月   7 19:30 buildroot/
drwxrwxr-x  2 brs brs 4096 9月   7 19:01 CI/
-rwxrwxr-x  1 brs brs  791 9月   7 19:01 download_and_test.sh*
drwxrwxr-x  9 brs brs 4096 9月   7 19:55 .git/
-rw-rw-r--  1 brs brs  288 9月   7 19:01 .gitignore
-rw-rw-r--  1 brs brs  420 9月   7 19:01 .gitmodules
drwxrwxr-x  7 brs brs 4096 9月   7 19:30 hdl/
-rw-rw-r--  1 brs brs 3868 9月   7 19:01 LICENSE.md
drwxrwxr-x 27 brs brs 4096 9月   7 19:30 linux/
-rw-rw-r--  1 brs brs 8014 9月   7 19:01 Makefile
-rw-rw-r--  1 brs brs 6423 9月   7 19:01 README.md
drwxrwxr-x  2 brs brs 4096 9月   7 19:01 scripts/
-rwxrwxr-x  1 brs brs 1495 9月   7 19:01 setup_env.sh*
drwxrwxr-x 23 brs brs 4096 9月   7 19:30 u-boot-xlnx/
brs@ubuntu-18:/opt/project/plutosdr-fw_0.38_libre$ 

Best Regards!