hz12opensource/libresdr #8: Changing Pre-Built Firmware Releases

TechMindsYT, 2024-05-17:
Hi Guys,

I successfully loaded from SD card the  baseclock pre-built firmware, but I would like to test the overclock versions too.

How can this be done?

I have tried copying the overclocked files to an sd card and inserting into the LibreSDR, but the LibreSDR does not boot now with SD Card. It will only boot without SD Card plugged in.

I had thought I could copy the pluto.frm file, but they don't exist in the pre-built folders.

*** Second cheeky question, how do we expand the frequency range? I tried the same ssh commands as original pluto, but it didn't work.

PS. I am making a tutorial video for this product and your firmware, so it would be awesome to get an early response.

Thanks,
Matt
(Tech Minds on YT)

hz12opensource, 2024-05-19:
Hi Matt, thanks for testing this firmware! So far I've only tried to boot the firmware from SD card and not the internal NAND flash. So you will need to copy the overclocked firmware to SD card and make sure it boot from SD card.

To try different overclocking options, you just need to unzip the [firmware packages](https://github.com/hz12opensource/libresdr/releases) to your SD card. There are different packages with different overclocks on that page. In my experience, my SDR runs pretty stable even with the highest clock posted there. Make sure to not boot from NAND flash if you have firmware previously installed to it.

To verify whether you have a correct overclock you can ssh into the SDR and `cat /proc/cpuinfo` and check the BogoMIPS (which should be half of clock frequency). If it fails to boot, taking a look at the serial console can be helpful.

If you are using the firmware built here the frequency range has already been expanded - no other steps are needed.

If you still encounter other problems I would recommend erase your current firmware on the NAND to ensure everything is from the SD card. I still need some time to test loading overclocked firmware to NAND (it should work if I rebuild the project from scratch, but I currently don't have time to test it).

Thanks and let me know if you have other questions.

TechMindsYT, 2024-05-19:
Thanks for the reply! :-)

Could you tell me how to erase the current firmware on the NAND?

I am also playing with compiling my own firmware. Which I managed to do and got it loaded. Now for some reason it wont write again to the flash even though I am following the same steps.

Maybe an erase of the NAND is required. but how to do this? :-)

Thanks
Matt

TechMindsYT, 2024-05-19:
I think I am a little confused at the procedure to either run from SD Card only, or to write to flash so no SD Card is required. I am not sure at what point it writes to flash..

Also my libre.frm from self compiling comes out at around 16Mb, where as others that I have seen are around 11 Mb. Weird. 

I will continue testing..

hz12opensource, 2024-05-24:
The NAND flash is not written (still contains your old firmware) unless you explicitly run an update script or using the USB interface to update it. 

I believe the NAND memory on the SDR is presented as /dev/mtd devices, and you can use relevant mtd commands to erase them.

The larger libre.frm is normal because this project enables compiler flags optimized for performance rather than size.

oborot-bolta, 2024-06-20:
> I updated the firmware using one of the libre.frm files linked in one of these issues and now the frequency expansion works perfectly, weird!

Where did you find libre.frm?

oddb1ts, 2024-09-16:
Hi i know this may be an old post but how do you clear / erase the nand ? , want to boot from card but it will not let me , and i can not get ethernet , so need to boot from card to see if i can fix this ? any help would be great , many thanks.

DG9BFC, 2025-03-20:
i used the overclock on sdcard .. and installed tezuka fw in nand
now i can swuitch between fw with plugging in the card (overclocked) or booting from nand

povnz, 2025-03-31:
> i used the overclock on sdcard .. and installed tezuka fw in nand now i can swuitch between fw with plugging in the card (overclocked) or booting from nand

could you share your method, I'm having issues getting the NAND flashed with Tezuka, I copy the 2 files but the lights never flash, I think I made the mistake of flashing OG PlutoFW rather than Libre, SD boot is fine

povnz, 2025-03-31:
> just like any pluto .. save the files and eject ... but only a few fw do "blinky blinky" with the libre .. so you just have to wait

Thank you so much, I had NO IDEA some FW didn't flashy flash.... I'm surprised I haven't bricked this thing .....

DG9BFC, 2025-07-01:
even if you "bricked" the unit (with writing firmware to flash that does not work) .. you can just throw a working fw on sd card and start from there