hz12opensource/libresdr #10: How to use PPS

wonshil, 2024-06-26:
How to use PPS?

Could I know some pps using example?

Pytlicek, 2024-10-04:
Not related to PPS, 
But I ordered 10MHz OXCO from Aliexpress: https://vi.aliexpress.com/item/1005007211198066.html
And MMCX male to SMA Male cable: https://vi.aliexpress.com/item/1005006716008066.html
On LibreSDR there is MMCX port on board.

[image]

Denisfk1985, 2024-10-04:
In the standard firmware of the manufacturer PPS and 10MHz interface it works. And also on VCTCXO half of the voltage and it gives a fairly accurate frequency. But in the current project and all that I came across, these functions do not work, they need to be implemented in HDL, for example, to adjust the frequency you need to supply voltage to VCTCXO, it comes from the DAC U9, which in turn is controlled by the SPI bus.

Pytlicek, 2024-10-04:
@Denisfk1985 "standard firmware" means from day0wl: https://github.com/day0wl/libresdr-fw ?

Denisfk1985, 2024-10-04:
I meant the firmware that comes with LibreSDR when purchased. I bought it a long time ago, and back then, LibreSDR was supplied with a pre-installed and pre-written microSD card. Now, the device is shipped with an empty memory card, but the seller provides a link to the firmware. I'm not sure what firmware is currently supplied; it’s probably the same as before, where the external synchronization ports and frequency correction for the crystal oscillator work correctly.

Shawn-McSorley, 2024-11-13:
Hi all,

I've made some progress here. The best place to look is at the [microphase ant sdr source code](https://github.com/MicroPhase/antsdr-fw). The libre sdr is likely a clone of the e310.

The main differences are the DAC chip and ports used by the FPGA. I've successfully modulated the frequency of the onboard VCTCXO.

The following constraints need to be added to the xdc file

```
set_property -dict {PACKAGE_PIN H18 IOSTANDARD LVCMOS33} [get_ports CLK_40M_DAC_nSYNC];
set_property -dict {PACKAGE_PIN F19 IOSTANDARD LVCMOS33} [get_ports CLK_40M_DAC_SCLK];
set_property -dict {PACKAGE_PIN F20 IOSTANDARD LVCMOS33} [get_ports CLK_40M_DAC_DIN];
set_property -dict {PACKAGE_PIN K18 IOSTANDARD LVCMOS33} [get_ports PPS_IN];
set_property -dict {PACKAGE_PIN H16 IOSTANDARD LVCMOS33} [get_ports CLK_40MHz_FPGA];
set_property -dict {PACKAGE_PIN K17 IOSTANDARD LVCMOS33} [get_ports CLKIN_10MHz]; # Same bank as CLK_40MHz_FPGA. 3.3V on data sheet.
```

The ports should also be created in the system_bd.tcl file

```
create_bd_port -dir I CLKIN_10MHz
create_bd_port -dir I CLK_40MHz_FPGA
create_bd_port -dir O CLK_40M_DAC_DIN
create_bd_port -dir O CLK_40M_DAC_SCLK
create_bd_port -dir O CLK_40M_DAC_nSYNC
create_bd_port -dir I PPS_IN
```

The ant sdr source code uses verilog modules made by Ettus research. The SPI driver code can be modified to support the chip on the libre sdr:

```
//
// Copyright 2015 Ettus Research, a National Instruments Company
//
// SPDX-License-Identifier: LGPL-3.0-or-later
//

`timescale 1ns / 1ps

// This code is intended for a dac5311 module (8-bit DAC). Though the data register should accompany the 12-bit version.
// Some notes from data sheet:
// 1. Max clock frequency of sclk is 20MHz when Vdd is 3.3V (see libre SDR schematic)
// 2. sclk always runs
// 3. See pg. 7 of data sheet for timing diagram. ~SYNC goes high. Data is clocked in on the next falling edge of sclk, after ~SYNC goes low.
// 4. 16 data bits are clocked in (16 sclk cycles afer ~SYNC goes low).
// 5. First two bits are operating mode. 00 is normal operation, 01 is 1kOhm to ground, 10 is 100kOhm to ground, 11 is high-z.
// 6. MSBs are clocked in first. Starting with operating mode, then 12-bit data. Rest is don't care.
// 7. ~SYNC must be low for at least 16 sclk cycles during write, otherwise data is ignored. 
// 8. ~SYNC must be high for at least 20ns before the next write. Falling edge of ~SYNC triggers write. 

// This code is a minor modification of ltc2630_spi.v by Ettus Research. 

module dacxx11_spi  (
    input   wire            clk,
    input   wire            rst,
    input   wire  [11:0]    data, // 12-bit data, 8 bits in MSBs
    output  reg             sclk,
    output  wire            mosi,
    output  reg             sync_n
);

//====================================================
//parameter define
//====================================================
localparam  IDLE        = 4'b0001;
localparam  SYNC_PRE    = 4'b0010;
localparam  DATA        = 4'b0100;
localparam  SYNC_END    = 4'b1000;

localparam NORMAL_OPERATION = 2'b00;
[... 117 lines cut ...]
        sclk <=  1'b0;
    end
end

assign mosi = data_shift[15];

//----------------sync_n------------------
always @(posedge clk ) begin
    if (rst==1'b1) begin
        sync_n <= 1'b1;
    end
    else if (state == SYNC_PRE && falling_edge == 1'b1) begin
        sync_n <= 1'b0;
    end
    else if (state == SYNC_END && rising_edge == 1'b1) begin
        sync_n <=  1'b1;
    end
end

endmodule
```

In the ant sdr source code, there's a module called axi_vcxo_ctrl which handles the 1PPS and 10MHz discipline logic. With the above modifications, this should be fairly easy to port to the libre sdr. I don't have a connector for the 1PPS or 10MHz port so can't check at the moment. I'll likely try discipline the VCTCXO off one of the RF inputs first.

I haven't exhaustively checked that the above code works. To test it I set one of the RF outputs to a constant frequency. I then used a phasemeter to measure its frequency, while changing the DAC input with a GPIO module. I was able to see the frequency of the RF output step up and down accordingly. If I make more progress here I may package it into new firmware.

tpwrules, 2024-12-03:
I have the right connector (MMCX e.g. https://www.digikey.com/en/products/detail/rf-solutions/CBA-SMAF-MMCXM/5845759)  and am interested in trying this.

sdy623, 2024-12-31:
So does anyone could make the LED light up, I found the output binded with LED0 ->J20 LED1 ->H21

sdy623, 2025-01-06:
I've made the **Renovation** of the shipped firmware with device. [To the commit](https://github.com/sdy623/hdl/commit/5b9bcef97dec9202b1c5d43871a1ca2c543c0460)

It seems that the shipped firmware doesn't support PPS input. Because I found the source hdl project of the shipped firmware. I updated the project with this repo's hdl commit with **Vivado 2022.2**. Actually, the user bought it from Taobao could get the orginial device developing package, a 22.5GB zip acrhive file called `ADI.zip`, but it was distributed with BaiduNetDisk and much 
of them are Chinese. In the acrhive, we could get the source code of the shipped firmware with  
the Vivado 2019.1 hdl project and SDK, we could build it up. It also provides the implmentent of
10Mhz REFCLK input, but the pps support doesn't covered. However, the **Vivado 2019.1** project may   
be old for developing, making a upgrade in its project could improve its performance a lot.

`ed407ace22fbfa8e07f1a5294c6234710a17721fe487ec0d8192f4e118d90044 ADI.zip`
I tested this project also with sdr++, and it made samping rate up to 24MSPS without overclocking.

F5OEO, 2025-01-11:
@sdy623 Congratulation for your Renovation hdl project. I will try to mix with my current implementation (including maia-sdr) at https://github.com/F5OEO/maia-sdr/tree/sweep/maia-hdl/projects/libre_iio. Could be amazing if you could contribute. Best !
Note that you can use https://github.com/F5OEO/tezuka_fw for building firmware instead of ADI toolchain.