hz12opensource/libresdr #11: No local ip context @192.168.2.1

maco326, 2024-06-30:
I finally got got my setup working to successfully build the project.  Right now, I am testing the device using the sdimg output, with plans to ultimately update the emmc...
When I connect the radio over ethernet, everything seems fine, but when I plug it in to USB, there is no local IP context.  Is this the expected behavior?  

maco@P65:~$ iio_info -s
Available contexts:
	0: (iwlwifi_1,nvme,BAT1,ADP1,coretemp,pch_cannonlake,nvme,acpitz on Micro-Star International Co., Ltd.) [local:]
	1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial= [usb:1.51.5]

While plugged in, my machine periodically notifies me that a network connection activation has failed.

Pytlicek, 2024-10-04:
The board I ordered with prebuilt image on card has default IP: 192.168.1.10 and when I set up my local LAN to static 192.168.1.1 - it worked without any issues. 
Strange is that over the LAN - I have got better performance output on SDR console than via the USB port.

magnum88, 2025-01-07:
For have USB network make it:
`vi /etc/init.d/S40network`
add this
`        echo "end 192.168.2.100" >> $UDHCPD_CONF`
in
```
         ### /etc/udhcpd.conf ###
        echo "start $IPADDR_HOST" > $UDHCPD_CONF
        echo "interface usb0" >> $UDHCPD_CONF
```

for have
```

         ### /etc/udhcpd.conf ###
        echo "start $IPADDR_HOST" > $UDHCPD_CONF
        echo "end 192.168.2.100" >> $UDHCPD_CONF
        echo "interface usb0" >> $UDHCPD_CONF
```

after save

put
`sh /etc/init.d/S40network restart`
and
`sh /etc/init.d/S41network restart`

try make one time  again
put
`sh /etc/init.d/S40network restart`
and
`sh /etc/init.d/S41network restart`

enjoy for have 192.168.2.1 usb

DG9BFC, 2025-07-01:
> The board I ordered with prebuilt image on card has default IP: 192.168.1.10 and when I set up my local LAN to static 192.168.1.1 - it worked without any issues. Strange is that over the LAN - I have got better performance output on SDR console than via the USB port.

sure ... ethernet supports up to 26 megs bandwidth .. way higher as via usb port