hz12opensource/libresdr #12: login fails with given firmware.

FPSychotic, 2024-07-01:
Hi, I bought this device. It has already flashed in internal memory the Pluto Firmware. The ADI chip is locked at 3.8ghz. Im trying connect via putty to unlock the 6ghz, and I need login.
Official firmware is root/analog, but doesnt work , tried analog/analog, and many others , but doesn't work.
Could you please tell me if it is the official firmware what is the password?
Thanks

maco326, 2024-07-04:
I just checked mine and it works with root/analog... 
How are you trying to connect? OTG port, DEBUG port or Ethernet connection?
root@192.168.2.1's password: analog
Welcome to:
______ _       _        _________________
| ___ \ |     | |      /  ___|  _  \ ___ \
| |_/ / |_   _| |_ ___ \ `--.| | | | |_/ /
|  __/| | | | | __/ _ \ `--. \ | | |    /
| |   | | |_| | || (_) /\__/ / |/ /| |\ \
\_|   |_|\__,_|\__\___/\____/|___/ \_| \_|

v0.37-dirty
https://wiki.analog.com/university/tools/pluto
#

FPSychotic, 2024-07-04:
Hi, thanks by answer, I do by OTG, I dont know setup ethernet  need find how, original pluto doesn't gave, so tried usn

maco326, 2024-07-04:
OTG with putty using ssh connection type should work.
DEBUG and putty using serial should work also with baud rate 115200 and correct com port selected.

I was not able to use Ethernet with factory firmware.  Using the pre-built firmware from here worked:
https://github.com/hz12opensource/libresdr/releases/download/v0.38/baseclock_cpu750_ddr525.tar.gz
This is booted from the sd card and does not change anything.  Remove the card and it boots back to factory firmware.

FPSychotic, 2024-07-08:
Hi thanks, via ssh worked, but the 6ghz hack didnt work in the stock PLutoSDR image in internal memory. not sure what is, maybe is different to other sellers

closed 2024-07-08 by FPSychotic