hz12opensource/libresdr #16: 1Gig Ethernet Port Not Working

wadetm, 2024-09-10:
I'm not able to get the LibreSDR to grab a DHCP IP when I plugin a ethernet cable into my network, is there a setting I need to change to enable the ethernet jack? 

BTW, I can access the LibreSDR over USB by setting a static IP on the Pluto Virtual Ethernet adapter, but it seems slow and I'm not able to get much use out of my intended use case with DATV-Express when using more than 2M BW channels for DVB-T.

wadetm, 2024-09-10:
I took board out of the aluminum enclosure and found that the ethernet transceiver being used is a RealTek  RTL8211E, does the firmware have drivers for this chip?

lucasmellon-ops, 2024-11-20:
Here's the workaround I used plug-in both USB C ports, wait. In your device manager You'll find the Pluto SDR serial console. Login username root, password analog. Type ifconfig USB0 down. Set a static address. Ignore what your router says, but in your software such as SDR sharp just type in the IP address you configured. I was pulling out my hair because my router would not recognize it being there but it's working.

DF3AK, 2024-12-01:
I have the same problem - the virtual USB ethernet interface works, but no dedicated ethernet. 

**Why would it be good to get the RTL8211E PHY running?**
I think it would be beneficial, because gigabit-ethernet  offers potentially more than twice the bandwidth (1000 MBit/s) when compared to USB2.0 (480 MBit/s). Both are lab-values of course, but gigabit ethernet clearly outperforms USB2.0. 
Further, if you would like to place your libreSDR close to the antenna, with the disadvantage of having some distance to your PC that might be too large for an USB connection, it would be much simpler to use an ethernet connection and USB-C power supply. 

**What is missing?**
It seems like we need to add another patch to the adalm-pluto fw repo. In fact, the pluto device tree source (zynq-pluto-sdr.dts, zynq-pluto-sdr.dtsi) does not contain an entry for PHY. Which totally makes sense, because the original pluto has no network interface. 

**How to?**
I am a total beginner in the Zynq/ARM world and thus I am not really sure. But what I figured out so far is:
- according to the schematics, the LibreSDR rev. 5 board comes with the RTL8211E-VB PHY
- the PHY is connected to Zynq7020 via PS_MIO16_501-PS_MIO27-501 and PS_MIO52_501-PS_MIO53_501. This seems to be compatible with, e.g., the zedturn-board for which we have 

**zynq-zed.dtsi**
```
&gem0 {
        phy-handle = <&phy0>;
        phy-mode = "rgmii-id";

        phy0: phy@0 {
                device_type = "ethernet-phy";
                reg = <0x0>;
                marvell,reg-init=<3 16 0xff00 0x1e 3 17 0xfff0 0x0a>;
        };
};
```
**zynq-zed.dts:** 
```
        aliases {
                ethernet0 = &gem0;
                ...
        };

&gem0 {
        status = "okay";
        phy-mode = "rgmii-id";
        phy-handle = <&ethernet_phy>;

        ethernet_phy: ethernet-phy@0 {
                reg = <0>;
                device_type = "ethernet-phy";
        };
};
```
- for the RTL8211E I found an [example](https://forum.digilent.com/topic/1294-zybo-ethernet-mdio-address-at-the-ref-manual/) of a device-tree source chunk. According to that, the **compatible** and **reg** fields should be set to 
```
                phy0: phy@1 {
                     compatible = "realtek,RTL8211E";
                     device_type = "ethernet-phy";
                     reg = <1>;
                };
```
But I am not sure about interrupts etc. 
- Finally, the stock FW, which was on the shipped SD-card, has ethernet working. But this FW seems quite different from the pluto FW. It looks like it is based on the ADI's Kuiper Linux image 2018_R2 and was thus not generated by following the instructions from the pluto-fw repo. Unfortunately, only the _BOOT.BIN_ and _image.ub_ are provided - I am uncertain about how to recover the device tree source files from that. 

**Next Steps**
I will do a bit reading on device tree files. Next, I will try to add the RTL8211E to the the current firmware source and see if I can get ethernet running. Might take some while, because my knowledge in this topic is rather limited.

DF3AK, 2024-12-01:
Problem solved, for me.

In fact, this repo already contains a device tree chunk:

```
+&gem0 {
+	status = "okay";
+	phy-mode = "rgmii-id";
+	phy-handle = <&phy0>;
+
+	phy0: phy@0 { /* Marvell 88e1512 */
+		reg = <0>;
+		compatible = "ethernet-phy-ieee802.3-c22";
+		reset-gpios = <&gpio0 46 1>;
+	};
+};
```

Apparently, the  _compatible_ and _reg_ fields are not that important. I have not tested the bandwidth, this would require a tool like perf on the libre. 

I had some issues with the SD-Card and it seems like I booted the wrong image. Now ethernet is working fine. The address can be configured in config.txt (virtual mass-storage) under the section **[USB_ETHERNET]**, after libreSDR is connected via USB-C (OTG plug).