hz12opensource/libresdr #19: Upgrade to v0.39

maggo1404, 2024-10-22:
Analog Device has released a new version 0.39.

VitGGGGGGGGGGGGGGGGGGGG, 2024-12-10:
please, make standart pluto firmware (for no SDCard mode) for Libre SDR with ethernet support!

Qw3rtzuiop, 2024-12-10:
> please, make standart pluto firmware (for no SDCard mode) for Libre SDR with ethernet support!

You can use the ethernet. Default ip is 192.168.1.10
Via usb connection and putty it is possible to set a new ip adress with the command: fw_setenv ipaddr_eth 192.168.178.100

VitGGGGGGGGGGGGGGGGGGGG, 2024-12-10:
Hi!
for me it is important to use the physical LAN interface as in pluto+. my LibreSDR should works remotely as a standard plutosdr.

Qw3rtzuiop, 2024-12-10:
> Hi! for me it is important to use the physical LAN interface as in pluto+. my LibreSDR should works remotely as a standard plutosdr.

Yes youre using the physical lan interface that way. i just needed to set the desired ip adress via usb first because my network isnt compatible with the default ip. After that you can reboot. Disconnect usb and plug in the ethernet connector.

VitGGGGGGGGGGGGGGGGGGGG, 2024-12-10:
How did you manage to do this? ifconfig shows that I have only one network interface 192.168.2.1. in the configuration file on the USB disk I specify the address for the physical LAN 192.168.137.100 but this does not lead to anything. for LibreSDR I found only modified 0.37 firmware which even with standard Adalmpluto in the original version does not have LAN. My PlutoPlus started working via physical lan only after installing modified firmware 0.38 with timestamp. I would like for LibreSDR the same firmware based on 0.38 where the network works without problems
[image]

Qw3rtzuiop, 2024-12-10:
> How did you manage to do this?
1.Copy firmware to sd card:https://github.com/hz12opensource/libresdr/releases
2. put in libresdr
3. connect to our computer via usb
4. open serial connection. user:root password:analog
5. change ip to your disired ip with fw_setenv ipaddr_eth 192.168.178.100
6. reboot with command reboot
7. success

VitGGGGGGGGGGGGGGGGGGGG, 2024-12-10:
but from the beginning I wrote that I don't need LibreSDR mode. I need this device to start up like Pluto WITHOUT AN SD CARD. this is necessary for compatibility with software that doesn't work with LibreSDR but with LAN support 
in other words i need pluto.frm file with 0.38 patched firmware so i can load it into SPI flash.

Qw3rtzuiop, 2024-12-10:
Then you are in the wrong repository.

VitGGGGGGGGGGGGGGGGGGGG, 2024-12-10:
I just wanted to suggest making libreSDR more functional for more applications.

VasylSamoilov, 2024-12-15:
Returning to original subject of the issue - will upgrade to 0.39 be possible? willing to test.

namus80, 2024-12-16:
Hello, can the same patches in this repository be used with the 0.39 firmware? Sorry I am complete noobie but happy to give it a try compiling in Ubuntu.

VitGGGGGGGGGGGGGGGGGGGG, 2024-12-16:
Most likely not, because the file structure in the new firmware may not fit the old patch.
 PS: By the way, about my question, after searching I found "0.37 dirty" firmware for plutosdr compatible mode that supports utp adapter. Now everything is fine.

VitGGGGGGGGGGGGGGGGGGGG, 2024-12-30:
it's because nicknames with less G were taken. and I don't want to change it just because of this

crackerjacques, 2024-12-30:
There is a universal firmware that is currently under development.
It is not yet compatible with LibreSDR officially, but the Issues page says he has bought it, so it is assumed that it will be supported in due course.
So far it is working fine in my Pluto+.

https://github.com/F5OEO/tezuka_fw
I wish there was a donation button to buy him a coffee to inject some motivation into him.

I like the name of this project because I read Tezuka so much as a kid that I memorised the lines.

DG9BFC, 2025-03-20:
Donate at https://www.paypal.com/paypalme/f5oeo

so there IS a donation link
dg9bfc sigi