hz12opensource/libresdr #22: Temperature moniting

VasylSamoilov, 2024-12-15:
Is there a way to monitor cpu and other relevant components temperatures? Maybe even some voltages, if sensors are present. 
Because while overclock works, it's hard to tell if we are withing temperature envelope, given that device can operate remotely, in different environments - it would be nice to have at least cpu temperature monitored.

DF3AK, 2025-01-08:
Yes, that is possible. We can read two temperatures remotely from the iio deamon.  Take a look into [here](https://github.com/analogdevicesinc/plutosdr_scripts/blob/master/pluto_temp.sh). 

E.g., if one calls 
```
iio_attr -u ip:[IP of the Libre] -c ad9361-phy temp0 input 
```
you get an integer. Divide it by 1000 and you get the current temperature of the AD9364.  

For the Zynq, we get a raw  value and two coefficients that form a linear equation. E.g., 
```
iio_attr -u ip:[IP of the Libre] -c xadc temp0 raw  # 2560
iio_attr -u ip:[IP of the Libre] -c xadc temp0 offset  # -2219
iio_attr -u ip:[IP of the Libre] -c xadc temp0 scale  # 123.040771484
```
The Zynq  temperature is then

(raw+offset)*scale/1000 = (2560-2219)*123.040771484/1000 = 41.96 °C (rounded).

Note that the latter seems to be the temperature of an ADC which is integrated into the Zynq's FPGA fabric, and therefore this is not actually the CPU temperature. But since both components are integrated into one chip, I would assume there is virtually no temperature difference.  

I guess both readings are uncalibrated, please consult the datasheets if calibration is relevant. Further, regarding the AD936X temperature, [this](https://ez.analog.com/cfs-file/__key/communityserver-wikis-components-files/00-00-00-02-20/AD936xTemperaturesensor.pdf) document states that 

"The temperature sensor has inaccuracies and even with a onetime calibration the sensor is typically not useful for accurate measurements."

How accurate is it? I do not know :) One could of course to some temperature ramps and observe the reading, but for most purposes this would be an overkill. 

To call iio_attr you will need to have libiio installed.

DG9BFC, 2025-03-20:
alberto (programmer of the well kown "satsagen" software (spectrum analyzer with tracking generator) ... made a temp readout software (after i told him that idea)
see here
https://www.albfer.com/wsdrtemp-download-page/
dg9bfc sigi