hz12opensource/libresdr #23: Second pair of antenas - how to use them? (in general and with SoapySDR)

VasylSamoilov, 2024-12-15:
As pluto have only one tx/rx, I have found no reasonable way to select rx2 or tx2 on libreSDR, or zynqSDR in my case. I presume second tx/rx share the same local oscillator like limeSDR, but it would be nice to utilize them anyway.
 
If I am missing something - any help is appreciated.

Here is the example from my SoapySDRUtil output:

```# SoapySDRUtil --probe="driver=plutosdr,hostname=192.168.160.10"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr,hostname=192.168.160.10
[INFO] Opening PlutoSDR #0 192.168.160.10...

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9361
  ad9361-phy,xo_correction=40000000
  backend_version=0.25 (git tag: v0.25)
  fw_version=v0.38-dirty
  hw_model=LibreSDR Rev.5 (Z7020-AD9361)
  hw_model_variant=1
  hw_serial=
  ip,ip-addr=192.168.160.10
  library_version=0.24 (git tag: v0.24)
  local,kernel=5.15.0-175882-ge14e351533f9-dirty
  uri=ip:192.168.160.10

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO
  Sensors: xadc_temp0, xadc_voltage0, xadc_voltage1, xadc_voltage2, xadc_voltage3, xadc_voltage4, xadc_voltage5, xadc_voltage6, xadc_voltage7, xadc_voltage8, adm1177_current0, adm1177_voltage0, ad9361-phy_temp0, ad9361-phy_voltage2
     * xadc_temp0: 43.802515 C
     * xadc_voltage0 (vccint): 0.974854 V
     * xadc_voltage1 (vccaux): 1.801758 V
     * xadc_voltage2 (vccbram): 0.972656 V
     * xadc_voltage3 (vccpint): 0.968994 V
     * xadc_voltage4 (vccpaux): 1.801758 V
     * xadc_voltage5 (vccoddr): 1.487549 V
     * xadc_voltage6 (vrefp): 1.254639 V
     * xadc_voltage7 (vrefn): 0.002197 V
     * xadc_voltage8: 0.002441 V
     * adm1177_current0:
     * adm1177_voltage0:
     * ad9361-phy_temp0: 48.246000 C
     * ad9361-phy_voltage2: 0.017399 V

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: A_BALANCED
  Full gain range: [0, 73] dB
    PGA gain range: [0, 73] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.260417, 61.44] MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=32768]
  Antennas: A
  Full gain range: [0, 89] dB
    PGA gain range: [0, 89] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.260417, 61.44] MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz
```

VasylSamoilov, 2024-12-15:
SoapyPlutoSDR has issue related to that: https://github.com/pothosware/SoapyPlutoSDR/issues/47 
In short, it should be compiled from branch to detect second tx/rx. Still untested.

DG9BFC, 2025-03-20:
tezuka fw supports it (but also most never"pluto" fw)

here a part of the config text from tezuka:

[TEZUKA]
callsign = NOCALL
locator = NOLOCATOR
lnb_power = off
refclk_source = internal
maxcpus = 2
attr_name = compatible
attr_val = ad9361
mode = 1r1t
audio_mode = off
rf_input = rx1
rf_output = tx1
serial_force = off
nfs_server = 
force_model = off
disable_usb_console = off
....................................
i guees if you change mode to 2r2t you can select what rx and tx port to be used
dg9bfc sigi