hz12opensource/libresdr #24: how to copy the build_sdimg to the sd card

gariac, 2024-12-29:
I downloaded and tar -xvf the prebuilt binary. Is that a .img file?

4Z1ZV, 2025-01-10:
You just open the compressed file and copy ASIS the entire directories and files to the SD.

gariac, 2025-01-11:
[image]

Well this looks odd since I have gunzip files present. 

I formated the sd card as ext4. I did a cp -R build_img/* to the sdcard .

DG9BFC, 2025-03-20:
i just formatted sd card as fat32 and dropped the files there
dg9bfc sigi

gariac, 2025-03-20:
That is what I did too. ;-) 

I do have Pluto-fw running on the ZYNQSDR. Basically the same idea. Just copy the files and plug it in.

DG9BFC, 2025-03-20:
sidenote
if you write a "pluto" fw (as example tezuka fw by evariste) in the nand ... and have the overclocking fw on sd card you have some kind of "dualboot " option (plug sd card = overclock ... unplug sd-card = load whatever fw you installed in the device itself)
1100/750 did not boot properly (or to be precise not at all) so i used 1000/700
dg9bfc sigi

gariac, 2025-03-24:
Wow. I avoided overclocking because I assumed that was trouble.  Thanks for the tip and indicating which version worked.  It never occurred to me that other than the overclocking feature that the behavior would be any different.

gariac, 2025-03-25:
> sidenote if you write a "pluto" fw (as example tezuka fw by evariste) in the nand ... and have the overclocking fw on sd card you have some kind of "dualboot " option (plug sd card = overclock ... [...]

I loaded that version of the firmware. It seems to believe just like the default version. It does work. 

Regarding pluging and unpluging sd cards, the device as far as I can tell just boots from the SD card. You can remove the sd card and nothing changes. I put in a different sd card and nothing changed.

DG9BFC, 2025-03-26:
plug in a new card nothing changes .. but plug in and reboot and it does
dg9bfc sigi

gariac, 2025-03-30:
> plug in a new card nothing changes .. but plug in and reboot and it does dg9bfc sigi

Except my problem is I have a SD card for

https://github.com/F5OEO/tezuka_fw

that won't boot.  That was why I was hoping this dual boot would work.

libresdr boots without issues. I don't see two SDRs though.

DG9BFC, 2025-03-30:
i did not use tezuka on sd card (but i could ... there even exist an overclocked one .. just replace the boot bin in the sd card folder against one of the overclocked versions in the subfolder "overclocked" and rename it)
i used this "hz12opensource" on sd card to boot ... then loaded the boot frm and pluto frm from tezuka on the "harddrive" and eject ... the tezuka fw is then saved in the nand
if you then boot from the card you have the hz12opensource running ... and if card unplugged the tezuka boots from the nand

DG9BFC, 2025-03-30:
but then you just have the "baseclock"! ... overclocked tezuka only runs from card not from nand (no overclocked folder for boot frm and pluto frm)