hz12opensource/libresdr #25: DATV

magnum88, 2024-12-29:
Hi all
i check for my ZynqSDR
a possible for make run on QO100 by DATV
i seek rtmp or udp
but by obs not working on 0.37v
only voici SSB working
it's normal?
thank's all

4Z1ZV, 2025-01-10:
If I recall, you may need to install modified FW.
Some support both SSB and CW and there are some patches that enable easy control.

Done it years ago on my Pluto based station.

https://www.f5uii.net/en/transmit-datv-over-qo100-with-sdr-adalm-pluto-f5oeo-plutodvb/4/

magnum88, 2025-01-12:
Thank's for your reply
but not work for libresdr not pluto blue it's pluto silver in case metal
73
Xavier

VasylSamoilov, 2025-01-17:
See here: https://github.com/F5OEO/tezuka_fw/issues/34
and here: https://github.com/F5OEO/tezuka_fw/commit/3e3178f5657a19b3768ab9c8950d058f17a5ee19

4Z1ZV, 2025-01-22:
Just an update, I was able to stream DATV signal using DATV-Easy 3.08 with a working fw (not modified for DATV).

magnum88, 2025-01-22:
Thank's for replay
for me no work with DATV-Red
only work SDRAngel and only USB connection no ETH