hz12opensource/libresdr #29: LibreSDR no TX output!

raimund56, 2025-01-27:
Hello,
I testet my LibreSDR with SDR-Console Version 3.3 build 2848 Ethernet and USB connection
with following firmware. LibreSDR is booting from the SD-card.
1 -(SD-card JT 2024 02 03) was shipped with LibreSDR
2 - Tezuka-3c31  F5OEO
3 - overclock-cpu900_ddr650
RX, works well without any problems.
TX, no output signal, if i change frequency sometimes peak on correct frequency comes up to 0dBm and goes down immediatley.
In the Spectrum i see a signal -50dBm a few 10kHz below the target frequency. ALC-Meter of course remains at "0".
All tests on TX1.
Have you any idea what the problem could be.
Thank you in advance!
Raimund

F5OEO, 2025-01-28:
LibreSDR with tezuka and sdrconsole tx is working here. If you consider using tezuka, you can write an issue with details on https://github.com/F5OEO/tezuka_fw/issues

raimund56, 2025-02-05:
Hello,
I was able to solve the TX-problem with an external USB-soundcard. 
It seems  SDRC does't uses the internal mic input with sampling rate 48kHz.
Raimund   in3her

sv5byr, 2025-02-11:
Hello, Tezuka firmware 0.95 have problem to SSB TX. The output form is DSB not Single Side Band.